confidential-containers/td-shim #632: Request to support to locate TDX metadata for TDVF with td-shim-tee-info-hash tool

jiazhang0, 2023-12-11:
Currently, td-shim follows Intel TDX Virtual Firmware Design Guide (Section 11.1) to locate TDX metadata, while TDVF follows OVMF GUIDed Structure to locate TDX metadata.

Actually, TDVF uses the same TDX metadata with td-shim but they just have different method to locate it. So it makes sense to implement the locating method for TDVF in td-shim-tee-info-hash tool rather than writing a new tool for TDVF.

OuyangHang33, 2024-01-08:
I will check this one

OuyangHang33, 2024-01-16:
td-shim-tee-info-hash tool can support 2 locating method mentioned above, but it set metadata buffer size based on TD_SHIM_METADATA_SIZE, which is 0x1000. 
I get the length from TDVF descriptor structure and set the metadata buffer based on it. Now it can locate TDX metadata correctly.

PR #651 (2024-01-29): Get metadata buffer size from TDVF DESCRIPTOR
Fix: #632
closed 2024-01-29 by jyao1