confidential-containers/td-shim #633: Request to use td-payload-reference-calculator to calculate kernel, cmdline and initrd eventlog digests

jiazhang0, 2023-12-11:
td-payload-reference-calculator is a common tool to calculate the typical artifacts for TDX.

Currently, kernel, cmdline and initrd eventlog digests are required to be calculated and verified when using TDVF + QEMU Kernel Direct Boot for CoCo.

Note that kernel eventlog digest is pre-processed by QEMU Kernel Direct Boot logic prior to PE authenticode. Here is the pseudocode and qemu source code references:
```
+               // https://github.com/qemu/qemu/blob/9c74490bff6c8886a922008d0c9ce6cae70dd17e/hw/i386/x86.c#L999
+               puts("patching @0x210 (type_of_loader) to 0xb0 ...");
+               p[0x210] = 0xb0;
+
+               // https://github.com/qemu/qemu/blob/9c74490bff6c8886a922008d0c9ce6cae70dd17e/hw/i386/x86.c#L1003
+               puts("patching @0x211 (loadflags) to 0x81 ...");
+               p[0x211] = 0x81;
+
+               // https://github.com/qemu/qemu/blob/9c74490bff6c8886a922008d0c9ce6cae70dd17e/hw/i386/x86.c#L1004
+               puts("patching @0x224 (heap_end_ptr) to 0xfe00 ...");
+               *(uint16_t *)(p + 0x224) = 0xfe00;
+
+               // https://github.com/qemu/qemu/blob/9c74490bff6c8886a922008d0c9ce6cae70dd17e/hw/i386/x86.c#L962
+               puts("patching @0x228 (cmd_line_ptr) to 0x20000");
+               *(uint32_t *)(p + 0x228) = 0x20000;
```

where `p` is the pointer to the base address of kernel image.

jyao1, 2023-12-11:
I think we may need two different mode for td-payload-reference-calculator.
1) Non-UEFI based, such as Td-shim
2) UEFI based, such as TDVF/OVMF

OuyangHang33, 2024-01-08:
I will check this one

PR #655 (2024-03-08): Add support for caculating kernel eventlog digist when using OVMF+QEMU
Fix: #633 

Tested with local bzImage binary and the result is matched with the programe output mentioned in #633 .
closed 2024-03-08 by jyao1