confidential-containers/td-shim #646: Debug binaries no longer fit into IPL section

jinankjain, 2023-12-29:
❯ cargo run -p td-shim-tools --bin td-shim-ld --features=linker -- target/x86_64-unknown-none/debug/ResetVector.bin target/x86_64-unknown-none/debug/td-shim -o target/debug/final.bin                              
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s                                                                                                                                                       
     Running `target/debug/td-shim-ld target/x86_64-unknown-none/debug/ResetVector.bin target/x86_64-unknown-none/debug/td-shim -o target/debug/final.bin`                                                          
[2023-12-29T11:45:49Z ERROR td_shim_tools] Size of IPL file (7988168) is invalid, should be in range [0-3440452]                                                                                                    
Error: Custom { kind: Other, error: "invalid file size" }

gaojiaqi7, 2024-01-04:
Hi @jinankjain 

I tried `cargo image` command which is used to build image with debug td-shim and can't reproduce the failure.

I assume you are using `dev` profile to build the td-shim? Please use `dev-opt` profile for debug td-shim because we need additional optimizations upon `dev` profile for binary size:

```
cargo xbuild -p td-shim --target x86_64-unknown-none --no-default-features --features main,tdx --profile dev-opt
```

jinankjain, 2024-01-04:
Correct it seems like the instructions in the cloud hypervisor docs seems outdated. I have a raised a PR to fix it. https://github.com/cloud-hypervisor/cloud-hypervisor/pull/6077

closed 2024-01-12 by jinankjain