confidential-containers/td-shim #654: [td-shim] publish td-shim in crate.io

NK-fengsf, 2024-02-01:
I am writing to ask if there are plans to publish td-shim in crate.io? 

Currently, td-shim is still in development, but it can already meet the basic usage needs. I am considering using td-shim in my project, but I need to know if there are plans to release an official crate for td-shim.

If there are plans , please let me know the specific release time and features.

Thank you!

NK-fengsf, 2024-02-01:
@gaojiaqi7  @jyao1

jyao1, 2024-02-21:
I think we just need to post the td-shim metadata related crate to crate.io, in order to let VMM consume it, right?

NK-fengsf, 2024-02-22:
> I think we just need to post the td-shim metadata related crate to crate.io, in order to let VMM consume it, right?
We need  data structure definitions of  both the TD-shim metadata and TD HOB.   The td-shim parse function in td-shim-tools crate are also needed.

studychao, 2024-02-22:
@jyao1 hi, we have updated the issue description to include all the crate we need in td-shim, please take a look.

OuyangHang33, 2024-02-26:
I will check this issue

studychao, 2024-03-04:
Hi @jyao1 @OuyangHang33 , although this issue is closed, but I couldn't find related crates in the `crate.io`. I have tried to search "td-shim" or "td-shim-interface" but I found nothing.

What's the name for the crate that is published? Thanks!

OuyangHang33, 2024-03-04:
Hi @studychao , we plan to publish crate after #657 tested and merged. I think we need reopen it.
Hi @jyao1 , please correct me if I'm wrong.

closed 2024-03-07 by jyao1

OuyangHang33, 2024-03-08:
td-shim-interface url: https://crates.io/crates/td-shim-interface

PR #660 (2024-03-04): Split file operation from loader.rs and change parse() API
Fix: #654

Splited the file operation and std env from loader.rs. Moved the read file operation to read_file.rs.

Test the final.bin with td-shim-checker tool and it can get the right result.