confidential-containers/td-shim #656: qemu: unsupported section type 5

mythi, 2024-02-26:
This repo has scripts that use QEMU so I gave `td-shim` a try using Canonical's TDX release. With that' QEMU errors:

```
qemu-system-x86_64: TDVF contains unsupported section type 5
qemu-system-x86_64: failed to parse TDVF for TDX VM
```

Any pointers to patches that are needed to get rid of this error?

gaojiaqi7, 2024-03-03:
Hi @mythi, thanks for your feedback! Currently `td-shim` can't work with qemu for Linux boot which relies on the `section type 5`, you can use [Cloud Hypervisor](https://github.com/cloud-hypervisor/cloud-hypervisor) instead.

mythi, 2024-03-05:
> Hi @mythi, thanks for your feedback! Currently `td-shim` can't work with qemu for Linux boot which relies on the `section type 5`, you can use [Cloud Hypervisor](https://github.com/cloud-hypervisor/cloud-hypervisor) instead.

OK, thanks! The qemu part is not documented very well. I could try cloud-hypervisor once TDX works with it.

haoyouab, 2025-03-20:
Hi. Since it's 2025 now, I wonder if Qemu has support for directly loading kernel with TD-Shim now. I checked the upstream qemu repo but didn't see any support. Is it already supported downstream?