confidential-containers/td-shim #666: Advisories detected by Cargo deny need to be resolved

liuw1, 2024-03-07:
2024-03-06 22:30:44 [WARN] unable to find a config path, falling back to default config
error[unmaintained]: json is unmaintained
   ┌─ /github/workspace/Cargo.lock:65:1
   │
65 │ json 0.12.4 registry+https://github.com/rust-lang/crates.io-index
   │ ----------------------------------------------------------------- unmaintained advisory detected
   │
   = ID: RUSTSEC-2022-0081
   = Advisory: https://rustsec.org/advisories/RUSTSEC-2022-0081
   = Last release was almost 3 years ago.
     
     The maintainer is unresponsive with outstanding issues.
     
     One of the outstanding issues include [a possible soundness issue](https://github.com/maciejhirsz/json-rust/issues/1[9](https://github.com/confidential-containers/td-shim/actions/runs/8179697021/job/22366243653?pr=659#step:5:10)6).
     
     ## Possible Alternative(s)
     
     The below list has not been vetted in any way and may or may not contain alternatives;
     
     - [jzon](https://crates.io/crates/jzon) maintained fork of json
     - [serde_json](https://crates.io/crates/serde_json)
     - [json-deserializer](https://crates.io/crates/json-deserializer)
     - [simd-json](https://crates.io/crates/simd-json)
   = Announcement: https://github.com/maciejhirsz/json-rust/issues/205
   = Solution: No safe upgrade is available!
   = json v0.12.4
     ├── bootloader-locator v0.0.4
     │   └── test-runner-server v0.1.0
     └── locate-cargo-manifest v0.2.2
         └── test-runner-server v0.1.0 (*)

warning[yanked]: detected yanked crate (try `cargo update -p iana-time-zone`)
   ┌─ /github/workspace/Cargo.lock:59:1
   │
59 │ iana-time-zone 0.1.59 registry+https://github.com/rust-lang/crates.io-index
   │ --------------------------------------------------------------------------- yanked version
   │
   = iana-time-zone v0.1.59
     └── chrono v0.4.31
         ├── chrono-tz v0.8.5
         │   └── tera v1.[19](https://github.com/confidential-containers/td-shim/actions/runs/8179697021/job/22366243653?pr=659#step:5:20).1
         │       └── td-layout-config v0.1.0
         └── tera v1.19.1 (*)

advisories FAILED

gaojiaqi7, 2024-03-07:
The `json` is used by both `bootloader-locator` and  `locate-cargo-manifest` which are referenced by `devtools/test-runner-server`. As `bootloader-locator` and  `locate-cargo-manifest` have not been updated for a while, we have two solutions:
1. Implement `locate_manifest()` and `locate_bootloader()` in `test-runner-server` crate without using third party library.
2. Find substitutions for these two crates.

I just tried to find out the replacement in crates.io but failed, so I prefer option 1. Any suggestions @liuw1 @jyao1

OuyangHang33, 2024-03-08:
I will check this

closed 2024-03-13 by jyao1