confidential-containers/td-shim #683: tee-info-hash tool calculated MRTD value mismatch with td_report for Qemu version 8.2.50

arronwy, 2024-04-01:
we tried with different guest kernel, OVMF, Qemu, and found `tee-info-hash` works with Qemu version 7.2.0 but failed with 8.2.50 @OuyangHang33

Qemu build command:
```
$ git clone --branch tdx-qemu-upstream-v5  https://github.com/intel/qemu-tdx
$ cd qemu-tdx && mkdir build; cd build
$ ../configure --target-list="x86_64-softmmu" --enable-kvm --prefix=XXX
$ make & make install
```
Qemu version:
```
/usr/local/bin/qemu-system-x86_64 --version
QEMU emulator version 8.2.50
```

td_report:
```
00000210  B4 81 1F 99 7A 20 F1 12 AD 9E 85 01 C9 DE C2 EE  ....z ..........
00000220  E0 00 8E B3 EC AF B5 8A D4 92 59 F5 DB 78 66 1B  ..........Y..xf.
00000230  EA 76 F5 CE 0C 52 DF 1A A8 7A F6 83 FF 47 F8 6B  .v...R...z...G.k
00000240  01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF  .#Eg.....#Eg....
00000250  01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF  .#Eg.....#Eg....
00000260  01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF  .#Eg.....#Eg....
00000270  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
```
MRTD calculated by `tee-info-hash`
```
./target/release/td-shim-tee-info-hash --manifest ./sample_manifest.json --image ~/tdx_vm/OVMF.fd  --seperator 0
[2024-04-01T08:58:24Z INFO  td_shim_tee_info_hash] TdInfo:
        Attributes:
        [0, 0, 0, 10, 0, 0, 0, 0]
        xfam:
        [e7, 2, 6, 0, 0, 0, 0, 0]
        MR TD:
        [64, 73, 65, f, a3, c4, 94, 5f, bd, 81, b4, 8c, 24, 74, 70, 12, ba, 5e, 71, e2, f5, 54, 76, 85, 32, 1e, 32, 8d, 73, c9, b, 1d, 6c, 8c, 3a, f7, 93, 69, 48, 73, 2c, fb, b6, 85, 49, 59, cc, 6f]
        MR Config ID:
        [1, 23, 45, 67, 89, ab, cd, ef, 1, 23, 45, 67, 89, ab, cd, ef, 1, 23, 45, 67, 89, ab, cd, ef, 1, 23, 45, 67, 89, ab, cd, ef, 1, 23, 45, 67, 89, ab, cd, ef, 1, 23, 45, 67, 89, ab, cd, ef]
        MR Owner:
        [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
        MR Owner Config:
        [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
        RTMR[0]:
        [51, 89, 23, b0, f9, 55, d0, 8d, a0, 77, c9, 6a, ab, a5, 22, b9, de, ce, de, 61, c5, 99, ce, a6, c4, 18, 89, cf, be, a4, ae, 4d, 50, 52, 9d, 96, fe, 4d, 1a, fd, af, b6, 5e, 7f, 95, bf, 23, c4]
        RTMR[1]:
        [51, 89, 23, b0, f9, 55, d0, 8d, a0, 77, c9, 6a, ab, a5, 22, b9, de, ce, de, 61, c5, 99, ce, a6, c4, 18, 89, cf, be, a4, ae, 4d, 50, 52, 9d, 96, fe, 4d, 1a, fd, af, b6, 5e, 7f, 95, bf, 23, c4]
        RTMR[2]:
        [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
        RTMR[3]:
        [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
```

OuyangHang33, 2024-04-03:
I sumary 2 scenarios here. Using Same OVMF.bin:
|Qemu        | OVMF.bin  | Guest Kernel | MRTD     |
| ----------  | -----------  |-----------      |-----------| 
| 7.2.0         | same          |  6.6     |  6473650fa3c4945fbd81b48c24747012ba5e71e2f5547685321e328d73c90b1d6c8c3af7936948732cfbb6854959cc6f
| 7.2.0         | same          |  6.8     |6473650fa3c4945fbd81b48c24747012ba5e71e2f5547685321e328d73c90b1d6c8c3af7936948732cfbb6854959cc6f
| 8.2.50       | same          |  6.6     | b4811f997a20f112ad9e8501c9dec2eee0008eb3ecafb58ad49259f5db78661bea76f5ce0c52df1aa87af683ff47f86b
| 8.2.50       | same          |  6.8     | b4811f997a20f112ad9e8501c9dec2eee0008eb3ecafb58ad49259f5db78661bea76f5ce0c52df1aa87af683ff47f86b

MRTD caculated from tee-info-hash: 6473650fa3c4945fbd81b48c24747012ba5e71e2f5547685321e328d73c90b1d6c8c3af7936948732cfbb6854959cc6f

OuyangHang33, 2024-04-08:
Tested with qemu 8.1.0, result is matched.

Regarding to 8.2.50, we tried to read MRTD in Shell without `kernel, device, append` paraments. The result was still dismatched.

OuyangHang33, 2024-04-09:
Hi @arronwy , we suppose that this issue is caused by tdx-qemu, if still have some questions, we can submit this issue to tdx-qemu :)

arronwy, 2024-04-09:
https://github.com/intel/qemu-tdx/issues/1

jyao1, 2024-05-31:
Since this is confirmed as QEMU-TDX issue, we can close this one.

closed 2024-05-31 by jyao1