confidential-containers/td-shim #715: Due to rsvd changed from '[u64; 3]` to `[u32; 5], build Test Image with Test TD Payload fail

OuyangHang33, 2024-06-18:
When building Test Image with Test TD Payload

Build test TD payload
```
$ cd tests
$ cargo xbuild -p test-td-payload --target x86_64-unknown-none --release --features=main,tdx
$ cd ..
```
error[E0277]: can't compare `[u64; 3]` with `[u32; 5]`
   --> tests/test-td-payload/src/testtdinfo.rs:101:32
    |
101 |         if (self.expected.rsvd != td_info.rsvd) {
    |                                ^^ no implementation for `[u64; 3] == [u32; 5]`
    |

jyao1, 2024-06-18:
I think we need add this to CI, to catch issue earlier.

liuw1, 2024-06-19:
> I think we need add this to CI, to catch issue earlier.
It's already added in CI. Since the integration test is blocked with the CI network issue, so CI didn't catch the issue. Currently, I have resolve the network issue, CI is recovered.

PR #717 (2024-06-20): Fix test payload build fail
Fix: #715
closed 2024-06-20 by jyao1