confidential-containers/td-shim #740: td-shim-tee-info-hash: MRTD calculation for OVMF

mythi, 2024-09-30:
### Describe the bug

I'm using `td-shim-tee-info-hash` to get OVMF generated MRTD "reproduced" but I'm not getting a match.

### How to reproduce

```
cargo build -p td-shim-tools --bin td-shim-tee-info-hash --features tee
target/debug/td-shim-tee-info-hash -i /usr/share/ovmf/OVMF.fd -m td-shim-tools/src/bin/td-shim-tee-info-hash/sample_manifest.json  -s 1 -o /tmp/foo.bin
```

The printed MRTD is `3491d438652cde331546683a37120504e961d02d871002f621fe51357df20c848406e485b625f2fd27bf3de32f49da70`.

My TDVM is booted with the same OVMF but the quote generated in it gives `91eb2b44d141d4ece09f0c75c2c53d247a3c68edd7fafe8a3520c942a604a407de03ae6dc5f87f27428b2538873118b7`

### CoCo version information

td-shim HEAD

### What TEE are you seeing the problem on

Tdx

### Failing command and relevant log output

```shell
$ ps ax|grep qemu
   7832 ?        Sl   11253:35 qemu-system-x86_64 -D /tmp/tdx-guest-td.log -accel kvm -m 2G -smp 16 -name td,process=td,debug-threads=on -cpu host -object {"qom-type":"tdx-guest","id":"tdx","quote-generation-socket":{"type": "vsock", "cid":"2","port":"4050"}} -machine q35,kernel_irqchip=split,confidential-guest-support=tdx,hpet=off -bios /usr/share/ovmf/OVMF.fd -nographic -daemonize -nodefaults -device virtio-net-pci,netdev=nic0_td -netdev user,id=nic0_td,hostfwd=tcp::10022-:22 -drive file=/home/mylinen/tdx/guest-tools/image/tdx-guest-ubuntu-24.04.qcow2,if=none,id=virtio-disk0 -device virtio-blk-pci,drive=virtio-disk0 -pidfile /tmp/tdx-demo-td-pid.pid
```

gaojiaqi7, 2024-12-20:
hi @mythi , I tried OVMF on my local machine and I can see the guest report MRTD matches the generated value using tee-info-hash tool. Could you share the tool command / manifest / OVMF image that can reproduce this issue?

mythi, 2024-12-20:
@gaojiaqi7 thanks for checking. Let me retry on my side to double check. Maybe I indeed had a user-error with the manifest (I remember I had some issues with it).

mythi, 2024-12-20:
> Could you share the manifest and OVMF image that can reproduce this issue?

I can reproduce the original issue still. I'm running Canonical 24.04 + TDX from it. `ovmf 2024.02-3+tdx1.0`. I don't see how the manifest file is relevant to `build_mrtd()`

gaojiaqi7, 2024-12-23:
I have tested both ovmf `edk2-stable202402` and `edk2-stable202411` with tdx1.5 and MRTDs are matched.

> I don't see how the manifest file is relevant to `build_mrtd()`
Right. The manifest affects `tee_info_hash` but not `mrtd`, I got them muddlled.

mythi, 2025-01-02:
> I have tested both ovmf `edk2-stable202402` and `edk2-stable202411` with tdx1.5 and MRTDs are matched.

I can try to get my setup updated for these and will get back to this. thanks!

jkr0103, 2025-02-04:
This issue seems duplicate of https://github.com/confidential-containers/td-shim/issues/683 which seem to highlight that qexu with tdx patches does the page processing it different way as mentioned [here](https://github.com/intel-staging/qemu-tdx/issues/1#issuecomment-2045259345). I created a patch which processes all the pages for a section first and than extends the section data in chunks which seem to give matching MRTD hash with qemu 9.0.2 from [canonical repo](https://github.com/canonical/tdx). I have attached the patch here in case someone want to try it.

Question to may be @jyao1, Is this tool indended only for qemu without tdx patches?

[tee_info_hash_mrtd_fix.patch](https://github.com/user-attachments/files/18655950/tee_info_hash_mrtd_fix.patch)

closed 2026-01-08 by jyao1